pnp-software/cordetfw #15: Type of CrFwPckt_t

pasetti, 2016-03-11:
On 18 Feb., Roland writes to say that, in the CORDET Framework, CrFwPckt_t is a char* (as defined in CrFramework/src/CrFwConstants.h). He thinks it should be a pointer to 'unsigned char' to avoid sign extensions.

pasetti, 2016-10-31:
Marcel: can you give me your assessment of this change? I believe that Roland is correct but the change may be tricky because it affects an external interface visible to framework users.

oppm, 2016-11-03:
Good point. If I am not mistaken, I think that the only place where the actual packet data are accessed is when you copy from/into the packet buffer using the function memcpy. memcpy does a cast to void* anyway, so from a cordetfw point of view it shouldn't matter if the CrFwPckt_t is char* or unsigned char* (or better: uint8_t*). But I actually tend to suggest changing it to void* to explicitly force the user to cast it to a reasonable type before accessing it.

pasetti, 2018-05-17:
I could make the change proposed by Roland (i.e. modifying `CrFwPckt_t `from "pointer to `char`" to "pointer to `unsigned char`") but I want to be sure that there are no repercussion on the code generated by the CORDET Editor. I have already verified that the generated code does not use `CrFwPckt_t `but I have also noticed that the packet accessor functions take a packet as an argument of type "pointer to void".

I will leave this change pending until after the integration of the CORDET Extension code to make sure that there are no interference on that side.