pnp-software/cordetfw #16: Coding Style

pasetti, 2016-03-25:
In Mantis Issue 705, Michael had proposed the following tool for enforcing/checking the coding style of the CORDET Framework:

[1] http://sourceforge.net/projects/astyle/ [^]

The following astyle command seems to match the style that is being used in the code.

astyle --style=java --indent=tab --indent-switches --unpad-paren --pad-header --keep-one-line-statements --keep-one-line-blocks --align-pointer=type --lineend=linux --suffix=none --quiet

eglimi, 2016-03-25:
There is now also ClangFormat, which might be a better option and gives more accurate results. Unlike astyle, it uses a real C parser (clang) to format the code.

http://clang.llvm.org/docs/ClangFormat.html

Other clang based tools might also be interesting for you, such as clang analyzer, which is a source code analysis tool.

http://clang-analyzer.llvm.org/index.html

cechticky, 2016-10-28:
I think we should go ClangFormat, clang tools in general ...

oppm, 2016-11-08:
I added a few thinks regarding coding style but on a separate branch for the fwprofile project. I just thought it is easier to start with the smaller of the two: https://github.com/pnp-software/fwprofile/tree/coding-style. The source code is formatted according to LLVM style. Have a look and comment.

In the project you can do:
1. make style - to format the source code according to the given stylesheet
2. make analyse - to run the source code analyser (finds the malloc/free problem)
3. make tidy - to check some other rules, e.g. for readability, best practises and the like

pasetti, 2018-05-17:
The link quoted by Marcel in his previous update no longer exists. Given that the coding style is uniform, I will just keep things as they are and close this ticket.

closed 2018-05-17 by pasetti