pnp-software/cordetfw #19: Wrong Type in Call to malloc

pasetti, 2016-10-23:
This point was first brought up by ESA when they did their static code analysis of the CORDET Framework code (see e-mail from Roland dated 11 October 2016).

In function `CrFwPcktQueueInit` of module `CrFwPcktQueue`, the following statement is made:

`pcktQueue->pckt = malloc(size*sizeof(CrFwPckt_t*));`

This statement might be incorrect and the correct statement is probably:

`pcktQueue->pckt = malloc(size*sizeof(CrFwPckt_t));`

I assign the ticket to Marcel so that he may check this matter and update the code if needed.

oppm, 2016-10-27:
Yes, it is wrong. I'll change it...

cechticky, 2016-10-28:
I saw that you solved the issue. Thanks, I'd suggest that you also close it.

oppm, 2016-10-28:
Someone needs to first merge the branch to the master branch.... And shouldn't the one close the issue who has opened it ?

cechticky, 2016-10-28:
I think it should be Alessandro, he is the maintainer for the cordetfw.

pasetti, 2016-10-31:
The problem has been fixed and the fix has been pulled into the master branch.

closed 2016-10-31 by pasetti