pnp-software/cordetfw #20: Dependency on fwprofile

oppm, 2016-10-27:
The current implementation assumes that fwprofile is located in a CrFramework folder just next to the cordetfw folder. This is not very nice and is not working with travis anyway. Unless we want to copy-paste the fwprofile files, we should link the two projects. This is done with submodules. I will create the accordant files and update the make- and header-files accordingly.

oppm, 2016-10-27:
It should build now. Including coverage export to codecov.io.

cechticky, 2016-10-28:
Personally, I'd remove the fwprofile folder. The cordetfw should reference fwprofile as an external library.

oppm, 2016-10-28:
fwprofile is now included as a submodule - it uses automatically the latest fwprofile version in the master branch. Thats why the folder is still there when you look at the branch I created, the submodules are shown automatically.

cechticky, 2016-10-28:
Understood. This should be fine then.

closed 2016-10-31 by oppm