pnp-software/cordetfw #37: Incorrect Definition of CrFwRepErrCode_t in Doxygen Documentation

pasetti, 2017-01-02:
The type `CrFwRepErrCode_t` defines an enumerated type with the list of error conditions handled by the framework. In the doxygen documentation, this enumerated type is shown as a table which list the values of the enumerated type but each value appears four times! This is probably because this type is defined in file CrFwUserConstant.h and there four instances of this file in the set of files covered by Doxygen (one for the test suite and three for the demo applications). 

The same problem occurs for other types which are defined multiple times.

If possible, we should try to avoid this duplication of information which is obviously confusing for the user.

oppm, 2017-01-03:
This seems to be a bug in doxygen. I found a few other reporting on this issue. It doesn't seem to be solved yet. I also did not find a suitable workaround, e.g. with using groups or the like. I don't really know what to do here.

pasetti, 2017-01-03:
If it is a doxygen bug, we will just have to live with it. I leave the ticket open just in case the issue is fixed in a future doxygen release.

pasetti, 2018-05-17:
The problem appears to have been fixed (I have used doxygen 1.8.11).

closed 2018-05-17 by pasetti