pnp-software/cordetfw #54: Proposal for Getter and Setter Functions for TM/TC Parameters

pasetti, 2017-07-24:
We have discussed how to set up the getter and setter functions for the TM/TC parameters. One objective is to make these functions very efficient. Based on various experiments done by Marcel, we think that one good approach is the one described below. Please let us know your feedback.

(a) For each TM/TC packet, a `struct` is generated which mimics the structure of the packet. Thus, for instance, if we had a packet with four items a, b, c and d of types: char, unsigned integer, array of chars, and unsigned integer, then the following structure would be generated:
```
    typedef struct __attribute__((packed, aligned(4))) _my_struct_t {
        uint8_t a;
        uint32_t b;
        uint8_t c[3];
        uint32_t d;
    } my_struct_t;
```
(b) For each parameter in a telecommand packet, a getter function would be generated to access the value of that parameter. In the case of the example, the functions generated for parameter A and B would look like this:

```
    uint8_t getA(my_struct_t* t)
    {
        return t->a;
    }

    uint32_t getB(my_struct_t* t)
    {
        return t->b;
    }
```

(c) For each parameter in a telemetry packet, a setter function would be generated to write the value that parameter. Its structure would be similar to that of the getter functions shown above.

(d) For parameters with an array-like structure in a telecommand, two getter functions would be generated: one which returns the array itself and another one which returns the value of the i-th element of the array. By default, no range checking will be implemented (it is therefore up to the user to make sure that the value of i is legal).

(e) For parameters with an array-like structure in a telemetry report, two functions would be generated: one setter function which to set the value of the i-th element of the array (without range-checking) and one getter function to return the array itself.

(f) For efficiency, all the getter and setter functions would be declared as inline functions (same approach as for the getter and setter functions of the data pool).

(f) For each service, a single header file would be generated which declares all the getter and setter functions for the commands and reports in that service.

The advantage of the approach outlined above is efficiency of implementation. Marcel has done some experiments using the code in the attached zip files and he finds that, in an optimal case, a getter or setter function is translated to one single assembler instruction (this is the case where the alignment of the parameter is just "right") whereas in a worst-case where some re-shuffling of data is needed to comply with alignment constraints, a getter/setter function is translated to four assembler instructions. For comparison, the approach which we used for CHEOPS required 11 assembler instructions for each getter/setter function.

The drawback of this approach is that it relies on the (non-standard) `__attribute__` compiler directive. This is generally undesirable but could be acceptable if the use of the directive is restricted to the generated code because users can chane the code generator to match their compiler.

pasetti, 2017-07-26:
### Summary of Teleconference of 26 July

The following points are agreed:

1. The access to the packet parameters will be through getter and setter functions (rather than through a direct access to data structures). This has the advantage that the access mechanism can be changed at a later time without impact on the user code.
2. The name of the getter and setter functions will be: get<PcktName><ParName>. We need to add the packet name to handle the case where the same parameter name is used in two different packets.
3. No range checking is needed for accesses to array elements: it user's responsibility to ensure that the array element is legal.

The following points are still open :

1. What should be the argument of the getter and setter functions? 
2. How do we handle parameters in repetition groups?
3. Do we need an initialization function which clears the packet? 

For decision point (1), there are two basic options:
(a) The argument of a getter/setter function is a pointer to a complete packet (header+body) seen as an array of bytes. This option is convenient for the user (because the framework infrastructure handles packets as arrays of bytes which include both the header and the body)  but hard to accommodate for the generated code.
(b)  The argument of a getter/setter function is a pointer to the body part of a packet seen as an instance of a `struct`.  This option requires the user to perform a cast from "array of bytes" to "pointer to `struct`" but it is more convenient for the generated code.

For the second decision point, the key issue is whether a range check should be done when accessing parameters in a repetition group. Note that this situation is different from the one concerning the access to the parameters in an array because, in the case of the repetition group, the range information (the number repetition count) is part of the packet being accessed whereas in the case of an array, the range information is not available within the packet. 

A simple and elegant way to implement the setter function for the members of a repetition group is to have an `add` function which allows all the parameters in one repetition group to be added to a packet. This approach only works if a packet is filled with "all-zeros" when it is allocated. Unfortunately, the framework infrastructure does not give any guarantee concerning the parameter values in a newly allocated packet. Hence, if we wanted to proceed with an `add` function, we would also need an initialization function which clears a newly allocated packet.

oppm, 2017-08-09:
I updated the datapool generator to also generate the getter/setter functions for the TC/TM packets. Just enable the datapool generator for a project and build it to get a taste of it.
An outline is given below:

`/**
 * Structure for packet Telecommand Acceptance Report - Success
 */
typedef struct __attribute__((packed, aligned(4))) _TelecommandAcceptanceReportSuccess_t {
   /**
    * Header spacer.
    */
   uint8_t __header[17];

   /**
    * This is a copy of the 16 bits of the TC Packet ID contained in the TC
    * Packet Header
    */
   uint16_t TCPacketID;
   
   /**
    * This is a copy of the 16 bits of the TC Packet Sequence Control
    * contained in the TC Packet Header
    */
   uint16_t TCPacketSequenceControl;
   
} TelecommandAcceptanceReportSuccess_t;

/**
 * Get "TCPacketID" from "Telecommand Acceptance Report - Success" packet.
 * \param p Pointer to the packet.
 * \return Value of "TCPacketID".
 */
static inline uint16_t getTelecommandAcceptanceReportSuccessTCPacketID(void* p)
{
   TelecommandAcceptanceReportSuccess_t* t;
   t = (TelecommandAcceptanceReportSuccess_t*)p;
   return t->TCPacketID;
}

/**
 * Set "TCPacketID" in "Telecommand Acceptance Report - Success" packet.
 * \param p Pointer to the packet.
 * \param TCPacketID Value of "TCPacketID" to be set in packet.
 */
static inline void setTelecommandAcceptanceReportSuccessTCPacketID(void* p, uint16_t TCPacketID)
{
   TelecommandAcceptanceReportSuccess_t* t;
   t = (TelecommandAcceptanceReportSuccess_t*)p;
   t->TCPacketID = TCPacketID;
}

/**
 * Get "TCPacketSequenceControl" from "Telecommand Acceptance Report -
 * Success" packet.
 * \param p Pointer to the packet.
 * \return Value of "TCPacketSequenceControl".
 */
static inline uint16_t getTelecommandAcceptanceReportSuccessTCPacketSequenceControl(void* p)
{
   TelecommandAcceptanceReportSuccess_t* t;
   t = (TelecommandAcceptanceReportSuccess_t*)p;
   return t->TCPacketSequenceControl;
}

/**
 * Set "TCPacketSequenceControl" in "Telecommand Acceptance Report - Success"
 * packet.
 * \param p Pointer to the packet.
 * \param TCPacketSequenceControl Value of "TCPacketSequenceControl" to be set
 * in packet.
 */
static inline void setTelecommandAcceptanceReportSuccessTCPacketSequenceControl(void* p, uint16_t TCPacketSequenceControl)
{
   TelecommandAcceptanceReportSuccess_t* t;
   t = (TelecommandAcceptanceReportSuccess_t*)p;
   t->TCPacketSequenceControl = TCPacketSequenceControl;
}
`

reimers, 2017-08-23:
The proposed getter and setter functions for the packet are generated and tested successfully with the testsuite. In our opinion we can close this issue.

closed 2017-08-23 by pasetti