pnp-software/cordetfw #55: Question on Definition Data Pool Items in PUS Extension Project

pasetti, 2017-07-24:
I am looking at the data pool definition in the CORDET Editor (PUS Extension Project) and I have the following questions:

(a) Why did you define user-specific types for unsigned char, unsigned short and unsigned integer? There are pre-defined types within the tool itself for these common types.

(b) Why is the default value of data items like nOfAccFailed or nOfPrgrFailed set to 1? Shouldn't the default value of these data items be zero?

reimers, 2017-07-24:
ad (a) 
if we use the  pre-defined types uint8_t, uint16_t and uint32_t we got the error "unknown data type" because the editor did not include the define of the stdint.h. This should be also included in the generated code.

ad (b)
this was only for testing purposes. But yes, you are right, for these type of data items (nOfXyz) the default values should be 0 instead of  1. We already changed this.

oppm, 2017-07-24:
(a) You can use -include stdint.h for gcc to include standard libraries.

reimers, 2017-08-11:
Thanks, we changed the make file and included the stdint.h (see new Version in github).

closed 2017-08-18 by pasetti