pnp-software/cordetfw #57: Implementation of Termination Action of (17,3)

pasetti, 2017-07-24:
The termination action of TC(17,3) is defined as follows in the PUS Extension Specification Document: "Set action outcome to 'success' if the (17,4) report was issued and to 'failure' otherwise". Its current implementation is as follows:

```
void CrPsTestOnBoardConnectionTerminationAction(FwSmDesc_t __attribute__((unused)) smDesc) 
{
  /* Set action outcome to 'success' */

  CrFwCmpData_t* inData;
 
  printf("CrPsTestOnBoardConnectionTerminationAction()\n");

  FwPrStop(prDescServTestOnBoardConnPrgr); /* TODO: to be investigated, why this is needed */

  inData = (CrFwCmpData_t*)FwSmGetData(smDesc);
  inData->outcome = 1;
 
  return;
}
```
This implementation is non-compliant to me because it always sets the action outcome to TRUE. Also, there should be no need for the ServTestOnBoardConnPrgr procedure to be stopped. This procedure terminates every time it is run. Hence, stopping it is unnecessary.

reimers, 2017-08-23:
The line with FwPrStop(prDescServTestOnBoardConnPrgr) have been deleted.

pasetti, 2017-08-23:
Thanks for removing the "Procedure Stop" statement. We still need to fix the logic to set the outcome. At present, this is always set to TRUE which, I believe, is non-compliant with the specification.

reimers, 2017-10-18:
Setting the outcome of the Termination Action is implemented using the outcome from the Progress Action. Thus should be compliant to the specifications.

closed 2017-11-08 by pasetti