pnp-software/cordetfw #58: Use of Non-Standard Compiler Directives in Framework Code

pasetti, 2017-07-24:
I notice that the code of the PUS Framework Extension makes extensive use of the __attribute__ compiler directive. This compiler directive is support by gcc but is not part of ANSI C. We should not use it in the manually-generated code because it limits portability!

As you have seen from a separate ticket, we are considering using for the automatically generated code. Even this kind of use needs to be carefully discussed and evaluated but *might* be acceptable because users have the option of changing the generator. However, the manually generated code cannot be modified by users because it is supposed to be pre-qualified. For this reason, only ANSI C should be used for that part of the code.

Rockenbauer, 2017-07-25:
We suggest to include the CR_UNUSED() function instead of the attribute compiler directive in the generated code. For example the service actions or the procedure nodes. This function call can be easily removed if the argument is used.

pasetti, 2017-07-26:
During the teleconference of 26 July, it is agreed that no compiler directives will be used in the manually generated part of the framework code.

closed 2017-07-26 by pasetti