pnp-software/cordetfw #60: Outcome of Start Action of TC(17,4)

pasetti, 2017-07-24:
I have looked at file CrPsCmd17s3StartFunc.c holding the implementation of the actions and guards of the Start Action of TC(17,4). Nodes N7 and N8 must report the outcome of the Start Action. At present, this is done using a global variable `outcomeStart` which, I believe, is defined in CrPsTestOnBoardConnection.c.

I believe that we should avoid using this kind of global variables because this usage is normally precluded by coding standard for critical software.

One alternative option could be to use the data field of the procedure descriptor

Rockenbauer, 2017-07-25:
We will implement the outcome handover using the data field of the procedure descriptor.

pasetti, 2017-07-26:
Thanks for the feedback. You may want to think about and define in your design document the general mechanism you plan to use for the management of the outcome of actions and checks.

I will leave the ticket open and assigned to you so that you may return it to me when you are finished with the update.

reimers, 2018-01-03:
The global variable <pre>outcomeStart</pre> is deleted and the approach using the data field of the procedure descriptor is implemented. The approach will be also defined in the design document.

closed 2018-01-21 by pasetti