pnp-software/cordetfw #62: Implementation of Start Action of TC(17,1)

pasetti, 2017-07-24:
The specification of the Start Action for TC(17,1) is: "Retrieve (17,2) report from OutFactory and set action outcome to success if retrieval succeeds. If the retrieval fails, generate error report OUTFACTORY_FAILED and set outcome of Start Action to failed". Its implementation is as follows:

```
void CrPsTestAreYouAliveConnectionStartAction(FwSmDesc_t smDesc)
{
  CrFwCmpData_t* inData;
  CrFwInCmdData_t* inSpecificData;
  CrFwPckt_t       inPckt;

  printf("CrPsTestAreYouAliveConnectionStartAction()\n");

  /* Get in packet */
  inData         = (CrFwCmpData_t*)FwSmGetData(smDesc);
  inSpecificData = (CrFwInCmdData_t*)inData->cmpSpecificData;
  inPckt         = inSpecificData->pckt;

  /* Send Request Verification Start Successful out-going report */
  SendReqVerifStartSuccRep(inPckt);

  inData->outcome = 1;

  return;
}
```
This implementation seems non-compliant in two respects:

(a) It does not retrieve the (17,2) report from the OutFactory and it always returns an outcome of success. Note that the retrieval of the (17,2) report is done by the Progress Action but there is no check on whether the retrieval was successful or not.

(b) It generates the "Start Successful" report for service 1. This is wrong because this report is generated by the CORDET infrastructure.

Rockenbauer, 2017-07-25:
ad (a) will be implemented as described in the PUS Extension Specification Document.

ad (b) the CrFwRepInCmdOutcome is an Adaptation Point and therefore have to be implemented by the application. Should this be implemented there?

pasetti, 2017-07-26:
Thanks for the update of the Start Action implementation.

> ad (b) the CrFwRepInCmdOutcome is an Adaptation Point and therefore have to be implemented by the application. Should this be implemented there?

Yes, indeed. By default, the CORDET Framework only offers stubs for the functions in interface CrFwRepInCmdOutcome. In your extension of the framework, you should provide the code which generates the service 1 responses.

I will leave this ticket assigned to you so that you may return it to me when the actions identified above have been completed.

pasetti, 2018-10-08:
The update and verification of the service 17 implementation has been completed. The ticket can be closed.

closed 2018-10-08 by pasetti