pnp-software/cordetfw #64: DatapoolId enumeration problem

Rockenbauer, 2017-07-25:
In the generated DataPool Header File (in our case:CrPsDp.h) where the datapool Id`s are defined the enumeration starts again from zero after the first two entries see sample code below.
The first two items are declared as Parameter in the CordetFw Editor all ohters are Variables.

```
typedef enum {
  DpIdAreYouAliveTimeOut = 0,
  DpIdOnBoardConnectDestLst = 1,
  DpIdfailCodeAccFailed = 0,
  DpIdfailCodePrgrFailed = 1,
  DpIdfailCodeStartFailed = 2,
  ....
```

oppm, 2017-07-25:
You can / have to set an manual offset in the Settings for the Standard. (Open the Communication Standard, click on "Settings...")

Rockenbauer, 2017-07-25:
OK i found the setting. But when i try to build there is an error: 'datapool' Something went wrong...
Even if i try to change the offset value back to zero i am not able to build anymore.
[image]

oppm, 2017-07-25:
Can you try to run a "Consistency Check" in menu "Edit" ? If it does not work, please export the project and send it to me (or upload here).

Rockenbauer, 2017-07-25:
Hallo!

 

Leider kommt auch hier eine Fehlermeldung: ‚datapool‘ Error: project not found.

 

Anbei der Export.

 

Vielen Dank

Markus

oppm, 2017-07-26:
Wo hast Du den export hingetan ??

Rockenbauer, 2017-07-26:
Sorry!

i thought i sent it to you as an E-Mail attachment. It is easier to put it here.

[20170725-2201__CrData.zip](https://github.com/pnp-software/cordetfw/files/1176439/20170725-2201__CrData.zip)

oppm, 2017-07-26:
Can you try again ? It should be ok now. I.e. I included now the consistency check as a prerequisite for the build process. So, you first have to fix the consistency issues before you can generate.

If it still doesn't work, I will add myself to the project so I can try online directly. Hope that is ok...

Rockenbauer, 2017-07-31:
I have tried, but the Problem stays the same. A Consistency check returns also an Error: 
```
'datapool'
Error: project not found.
```

Of course you can add yourself to the project.

pasetti, 2018-05-17:
I close this issue because the consistency check is now working correctly (at least on the CORDET FW Project). In any case, Marcel has left and hence, if there are problems, they will need to be documented through new issues.

closed 2018-05-17 by pasetti