pnp-software/cordetfw #65: Initialization of DataPool items

Rockenbauer, 2017-07-25:
in the current version of the DataPool initialization Travis shows following error:

`./src/DataPool/CrPsDpServReqVerif.c:21:7: error: ISO C90 forbids specifying subobject to initialize [-Werror=edantic]`

because it is not ISO C90 style. Instead of the generated code:
```
void initDpServReqVerif(void)
{
   dpServReqVerif = (DpServReqVerif_t){
      .failCodeAccFailed = 0,
         ...
      .pcktIdTermFailed = 0,
      .stepPrgrFailed = 0,
      .verFailData = 0
};
```
we tried to change the code to fulfill the ISO C90 style like:
```
void initDpServReqVerif(void)
{
      dpServReqVerif.failCodeAccFailed = 0;
      ...
      dpServReqVerif.pcktIdTermFailed = 0;
      dpServReqVerif.stepPrgrFailed = 0;
      dpServReqVerif.verFailData = 0;
}
```
Would it be possible to change the editor so the generated code is ISO C90 style?

oppm, 2017-07-25:
ok, I did so. You can try online, I updated the stuff.

Rockenbauer, 2017-08-03:
We tried to compile the code, but there where several problems.
We had to replace the commas at the end of each line with a semicolon.
Also at the end of the last line there should be a semicolon.
For the values with multiplicity > 1 the initialization with the {} curly brackets does not work.
Each array element should be initializes separately. see below:

```
void initDpServTest(void)
{
   dpServTest.AreYouAliveTimeOut = 10;
   dpServTest.OnBoardConnectDestLst[0] = 0;
   dpServTest.OnBoardConnectDestLst[1] = 0;
   dpServTest.OnBoardConnectDestLst[2] = 0;
   dpServTest.OnBoardConnectDestLst[3] = 0;
   dpServTest.OnBoardConnectDestLst[4] = 0;
   dpServTest.OnBoardConnectDestLst[5] = 0;
   dpServTest.OnBoardConnectDestLst[6] = 0;
   dpServTest.OnBoardConnectDestLst[7] = 0;
   dpServTest.OnBoardConnectDestLst[8] = 0;
   dpServTest.OnBoardConnectDestLst[9] = 0;
   dpServTest.AreYouAliveSrc = 0;
   dpServTest.OnBoardConnectDest = 0;
}
```

pasetti, 2018-10-08:
I close this issue because the data pool generated by the editor initializes and can be correctly compiled.

closed 2018-10-08 by pasetti