pnp-software/cordetfw #67: Doxygen Comments for Test Case Functions

pasetti, 2017-07-26:
Each unit-level test case is encapsulated in a function with a name like CrPsServVeriTestCase1. I would expect the corresponding test case to be defined in the doxygen comment for the test case itself. But the header file where the service 1 test cases are defined looks like this:

```
/**
 * Test the Service 17 Are-You-Alive Test Connection
 *
 * The following actions are performed in this test:
 * - TBW
 * .
 * @return true if the test was successful, false otherwise.
 */
CrFwBool_t CrPsServVeriTestCase1();
CrFwBool_t CrPsServVeriTestCase2();
CrFwBool_t CrPsServVeriTestCase3();
```
Unless you want to define all test cases in a separate document, it is mandatory to give the full definition in the doxygen comment of each test case. For reference, here is an example of the doxygen documentation of a test case function for the framework:

```
/**
 * Check the execution of a "pending" OutComponent with a valid destination.
 * The following actions are performed in this test:
 * - The OutFactory is reset and then one Sample1 OutComponent (see <code>CrFwOutCmpSample1.h</code>) is
 *   retrieved from the OutFactory.
 * - The Enable Check of the sample OutComponent is configured to return "enabled", the ready
 *   check is configured to return "not ready", and the repeat check is configured to return "no repeat".
 * - The Destination of the sample OutComponent is set to an invalid value (a value to which an
 *   OutStream is associated)
 * - The sample OutComponent is sent the Execute and Terminate commands and it is checked that
 *   it enters state PENDING.
 * - The sample OutComponent is sent the Execute and Terminate commands two more times and it is
 *   checked that it remains in state PENDING.
 * - The ready check is configured to return "ready" and then the sample OutComponent is sent the
 *   Execute and Terminate commands again.
 * - It is checked that the sample OutComponent is in state TERMINATED and that the OutComponent
 *   has been serialized to the OutStream
 * .
 * @verify OutComponent SM Transition: IPS->LOADED
 * @verify OutComponent SM Transition: LOADED->CPS1
 * @verify OutComponent SM Transition: CPS1->PENDING
 * @verify OutComponent SM Transition: PENDING->TERMINATED
 * @verify Send Packet Procedure: OutStreamRegistry returns an OutStream
 *
 * @return true if the test was successful, false otherwise.
 */
CrFwBool_t CrFwOutCmpTestCase5();
```
Note that, in the CORDET Framework Project, I used the doxygen tag "@verify" to more easily keep track of which items were covered by each test case. You may want consider a similar approach but the most important requirement is that the test case be clearly and unambiguously defined somewhere (either in a dedicated document or in the doxygen comments).

Rockenbauer, 2017-07-31:
Yes, i was writing the TestCases without comments at all. This was a big mistake! I am working on commenting the code as well as writing information into the header files. I will take the CORDET Framework Project TestCase header files as a template.

reimers, 2017-08-11:
We added further Information in the header files of the test cases (see new version in github).

closed 2017-08-18 by pasetti