pnp-software/cordetfw #68: Generate Error Report OUTFACTORY_FAIL

pasetti, 2017-07-26:
I have seen several "TODO" comments in the code covering the generation of the error report OUTFACTORY_FAIL. It is up to you how to handle error reports but you may want to follow the same approach as used in the CORDET Framework where interface `CrFwRepErr` defines a set of generic functions for reporting errors defined at framework level. You could define a similar interface for reporting errors defined at PUS level (such as OUTFACTORY_FAIL). 

For instance, for the case of the OUTFACTORY_FAIL error report, you would need a function which takes the error identifier and three parameters representing the type, sub-type and discriminant of the report which could not be created.

reimers, 2017-08-04:
interface CrPsRepErr to handle INLOADER_INV_DEST, INLOADER_ACC_FAIL and OUTFACTORY_FAIL errors is now implemented

pasetti, 2017-08-18:
Thanks for the update. By the way, the CrPsRepErr.h file includes the following text:

>  * In general, the implementation of this interface is entirely application-specific
>  * but a simple default implementation is provided in <code>CrFwRepErr.c</code>.
>  * This default implementation is primarily intended for testing and demonstration
>  * purposes.
>  *
>  * @author Vaclav Cechticky <vaclav.cechticky@pnp-software.com>
>  * @author Alessandro Pasetti <pasetti@pnp-software.com>
>  * @copyright P&P Software GmbH, 2013, All Rights Reserved

This is probably a cut-and-paste error and you may want to update it with a reference to CrPsRepErr.c and with your and Christian names as authors and the university as copyright holder

reimers, 2017-10-18:
The comments will be changed accordingly.