pnp-software/cordetfw #71: OutFactory CR_FW_OUTFACTORY_MAX_NOF_OUTCMP

Rockenbauer, 2017-08-08:
The value can not be greater then 10.
If we set it to a bigger value, no error occurs, but it is not possible to allocate more then 10 OutComponents.
If you want to allocate more than 10 OutComponents the function `CrFwOutFactoryGetNOfAllocatedOutCmp()` always returns 10.

pasetti, 2017-08-17:
I cannot reproduce this problem. In fact, I have tried the following:

- In the test suite of the CORDET Framework, I have modified the value of CR_FW_OUTFACTORY_MAX_NOF_OUTCMP in file tests/config/CrFwOutFactoryUserPar.h to 12
- I have run the test suite and I find that there are no errors
- As an extra check, I ran the test suite with the debugger and verified that 12 OutComponents are extracted from the OutFactory

Thus, it seems to me that all works as expected.

One possible problem is the following: the OutFactory only returns an OutComponent if function CrFwPcktMake is capable of generating a packet of sufficient length to hold the OutComponent. Are you sure that this function is configured to be able to generate more than 10 packets?

Does the same problem also exist in the CHEOPS application? I do not believe so because I suppose that Roland would have told me.

Rockenbauer, 2017-08-23:
Yes! All works fine now.
I downloaded a clean version of the CordetFramework, and set the `CR_FW_OUTFACTORY_MAX_NOF_OUTCMP` value to 12 and it wored fine. Then i tried setting the value to 13 and some of the OutFactory TestCases failed.
I realized that there is a `CR_FW_MAX_NOF_PCKTS` in the `CrFwConstants.h` that was set to 12.
So, my fault sorry. 
Of course i have to set this maximum number of packets higher than the maximum number of outcomponents.

closed 2017-08-23 by pasetti