pnp-software/cordetfw #77: Stopping Ready Action of Report HkRep

reimers, 2017-10-06:
The only stop condition the procedure which is runned by the Ready Check Action of Report HkRep (shown in the specification document in Fig. 9.5) is if the SID is undefined. But if the OutCmp of the HkRep will be disabled by setting the SID to undefined the Ready Check could never execute the procedure in the Ready Check Action in order to go to the final point. This should be done in HkDeleteCmd module.

pasetti, 2017-10-07:
While investigating this ticket, I realized that there is a flaw in the specification of some of the service 3 reports and commands.

**Preamble**
Reports of type (3,25) or (3,26) are OutComponents.

In the CORDET FW, when an OutComponent is executed, it performs an Enable Check (see figure 6.2 in the CORDET Definition Dcument). If the Enable Check returns "not enabled", the OutComponent is aborted and dies. 

**Enable Status of Service 3 Reports**
The specification of (3,25) and (3,26) reports associates to each SID in the RDL an "Enable Status" (flag `isEnabled` in table 9.1 in the PUS Extension Specification Document). This "Enable Status" refers to the housekeeping/diagnostic report being enabled or disabled and is therefore **not** the same as the Enable Status of a generic OutComponent! In my specification, I confused these two Enable Statuses and this resulted in a number of mistakes which I explain below.

**Mistake 1: Definition of Enable Check of HkRep Report (Table 9.8)**
The HkRep report encapsulates a (3,25) or a (3,26) report. This report is created by a HkCreateCmd command and is deleted by a HkDeleteCmd command. While it is alive, is generated periodically if it is enabled or else it is remains pending and is not generated. 

The Enable Check of the HkRep report is currently defined as: "_The enable status is read from the
`isEnabled` field of the Report Definition corresponding to the report's SID_". This definition is wrong: the isEnabled field in the RDL is used by the Ready Check to decide whether the report should be generated in the current cycle. 

The Enable Check of the HkRep report should be changed as follows: "_The report is enabled if its SID is defined in the RDL_". This definition ensures that the report remains alive until the HkDeleteCmd is received which removes its SID from the RDL.

**Ready Check of HkRep Report (Table 9.8**
After the correction described at the previous point, the Ready Check of HkRep (figure 9.5) can be seen to be correct. This can be understood by considering the life-cycle of a (3,25) report:

- When the report is created by a HkCreateCmd, it is loaded in its OutManager and is then periodically executed
- While the report is alive, it may enabled and disabled and, when it is enabled, its Ready Check will return "ready" once per period
- When the Ready Check returns Ready, the report is generated
- When the report is deleted by a HkDeleteCmd, its SID is removed from the RDL (see Progress Action in table 9.3). At this point in time, the report is still loaded in the OutManager
- When the OutManager is next executed, the Ready Check Procedure of the report is executed and this will find that the SID is not defined and the procedure will therefore terminate. Also, the Enable Check of the report will return "not enabled" and hence the HkRep will terminate and will be unloaded from the OutManager

reimers, 2017-10-18:
Right, we will change the implementation accordingly.

closed 2017-10-18 by reimers