pnp-software/cordetfw #78: Adding new constant for HkOneShot

reimers, 2017-10-06:
We decided to reserve a specific range of values for the SID which is used vor HkOneShot reports. This range should be defined in table 9.11 (Constants for Service 3) of the PUS Extension specification document.

pasetti, 2017-10-11:
As discussed today, I have modified the logic for the generation of one-shot reports as follows:

- The HkRepOneShot component has been deleted
- The period attribute in the RDL is now allowed to take on the value zero to represent a situation where a report is a one-shot report
- The progress action of the HkOneShotCmd command has been modified to just set the cycle counter of the target report equal to its period. This ensures that the report's Ready Check will return "ready" in the next cycle and this in turn implements the "one-shot release" of the report.

reimers, 2017-10-18:
We will check and implement the changes accordingly.

closed 2017-10-18 by reimers