pnp-software/cordetfw #82: Packets with dynamic length

reimers, 2017-10-10:
During the implementation of TC(3,1) Create a Housekeeping Parameter Report Structure we found that the functionality of the automatically generated getters and setter are not sufficient as it is implemented in the Cordet FW Editor yet. Because the length of some TC/TM packets is not fixed but have to be fixed within the current Cordet FW Editor. E.g. in TC(3,1) there is the parameter N1 which says how many parameter IDs will follow. To get the parameter NFA you have to now N1 which is individually set. The same accounts for all repeated structures in the data field of packets but especially in service 3 in the case of TC(3,1), TC(3,2), TM(3,10), and TM(3,12).

pasetti, 2017-10-10:
I agree with Christian's assessment. The problem can be illustrated for the case of TC(3,1) but applied more generally wherever there is a packet which has a group which is repeated N times and N is itself a parameter. The layout of the TC(3,1) is as follows (figure taken from PUS standard):

[image]

The problem here is that, in order to get the value of NFA, one needs to know the value of N1. For a getter function, this means that the function must be aware that NFA is located after a group and it must be aware of where, within the group, it can find the size of the group.

For a setter function, we must impose the constraint that the value of NFA is set after the value of N1 has been set (or, more generally, that the value of a parameter located after a group is only ever set after the value of the group size has been set).

I assign the ticket to Marcel to check whether the above logic can be implemented in the getter and setter functions.

pasetti, 2017-10-11:
After discussion with Marcel, we have come to the conclusions that it does not always make sense to provide getter and setter functions when a packet contains a group (i.e. a set of elements which are repeated N times where N is itself a packet parameter) or when it contains a sequence of parameters of types which are statically undefined. Below, I consider the various cases which can arise.

**Case 1: packet with one group with elements with a fixed size**
An example of this case is provided by TC(3,5):

[image]

Here, the group only has only one element with a fixed size and it is basically equivalent to an array. A slightly more complicated case is TC(3,31) where the group consists of two elements with fixed sizes:

[image]

In such cases, the generator currently provides getter and setter functions which allow the entire set of groups to be set and read as one single entity. Marcel will provide two extra functions which allow a specific parameter within a specific group element to be set and read.

**Case 2: packet with groups with non-fixed parameter sizes or with embedded groups**
An example of this case is the TC(3,1) shown at the top of this ticket. In such a case, a getter function would have to first parse the packet to identify the values of N1, NFA and N2 and would then have to use these values to identify the location of the desired parameter identifier. This task would have to be done each time the getter function is called. This is far less efficient than having dedicated code which parses the packet once and retrieves the parameter values in one single reading cycle. 

The situation is even more awkward for a setter function which can only set the value of a parameter identifier if the values of N1, NFA and N2 have already been set.

Hence, for such a case, no automatically generated getter and setter functions can be provided and the reading and writing of the parameters must be done by dedicated code.
 
**Case 3: packet with sequences of parameter of non-fixed size**
An example of this case is the (3,25) report:

[image]

In this case, getter and setter functions are provided for reports which are statically defined (i.e. for housekeeping reports which are defined in the CORDET Editor) but they cannot be provided for reports which are created dynamically using TC(3,1).

reimers, 2018-01-03:
Agreed, for Case 2 we implemented specific getter and setter functions.

closed 2018-01-03 by reimers