pnp-software/cordetfw #90: Multiplicity of Procedure for Ready Check of HkRep Component

pasetti, 2017-10-16:
As far as I can tell from the code, there is only one single instance of procedure prDescHkRepReadyCheck. This procedure runs for as long as a HK Report remains pending. The procedure needs to know upon which RDL slot it operates. At present, this knowledge is provided through the SID: the SID is stored in the prDataHkRepReadyCheck_t data structure and the procedure then uses the SID to find the RDL slot where the report holding that SID is stored (see for instance function CrPsRep3s25ReadyN1). This seems inefficient to me because it forces the procedure to search the entire RDL for a SID. A more efficient solution would be for the  prDataHkRepReadyCheck_t data structure to hold the RDL slot where a given SID is stored.

reimers, 2018-01-05:
Yes agreed, this is inefficient. We changed sid to rdlSlot in prDataHkRepReadyCheck_t data structure.

pasetti, 2019-03-19:
This issue has become obsolete further to the re-definition and re-implementation of service 3.

closed 2019-03-19 by pasetti