pnp-software/cordetfw #98: Creation of HK reports which are disabled

reimers, 2017-10-31:
We have found the problem, why newly created HK reports could not be loaded to run periodically the "Ready Check". This is described in the chapter 8.1.2 (The Command Conditional Checks) of the User Manual for the C2 Implementation (PP-UM-COR-00002):

**Ready Check**
This check is performed on a pending command instance that has passed
its enable check. The ready check determines when the command instance
is sent to its destination. The command instance remains pending until
the ready check is passed. When the ready check is passed, the command
instance may be sent to its destination.

**Repeat Check**
**This check is performed on a command instance after it has been sent
to its destination.** The check returns either "repeat" or "no repeat". In
the former case, the command instance is updated and sent again to its
destination. In the latter case, it is terminated.

According to the specifications the first run of the "Ready Check" results in "Not Ready" and therefore the "Repeat Check" is never called.

Some possible solution could be:

1. call the "Repeat Check" after the "Enable Check" or after each "Ready Check", which leads to changes in the CORDET framework

2. manually trigger the "Repeat Check"

3. to ensure that the first result of the "Ready Check" is "Ready", which means the execution of the "Update Action" and also the "Repeat Check". The disadvantage is here that at least one report is sent to the destination.

What do you mean?

pasetti, 2017-10-31:
Christian's analysis is correct: there is indeed a bug in the CORDET Framework implementation. 

Note that the specification of the CORDET Framework is correct. If one looks at figure 11.1 of the CORDET User Manual, one can see that flag isRepeat is set to TRUE when the command after the command has been loaded (see the action on the transition out of state LOADED). This should guarantee that a command which is not ready remains pending until it becomes ready. However in the implementation, there is an error in the Entry Action of the PENDING state. This action is implemented by function  PendingEntryAction of module CrFwOutCmp:

```
void PendingEntryAction(FwSmDesc_t smDesc) {
	CrFwCmpData_t* cmpData = (CrFwCmpData_t*)FwSmGetData(smDesc);
	CrFwOutCmpData_t* cmpSpecificData = (CrFwOutCmpData_t*)(cmpData->cmpSpecificData);
	CrFwBool_t isReady;
	CrFwBool_t isRepeat;
	CrFwDestSrc_t dest;
	FwSmDesc_t outStream;

	isReady = cmpSpecificData->isReady(smDesc);
	if (isReady) {
		cmpSpecificData->update(smDesc);
		dest = CrFwPcktGetDest(cmpSpecificData->pckt);
		outStream = CrFwOutStreamGet(dest);
		if (outStream == NULL) {
			CrFwRepErrDestSrc(crOutCmpSendPcktInvDest, cmpData->typeId, cmpData->instanceId, dest);
			isRepeat = 0;
		} else {
			CrFwOutCmpSetTimeStamp(smDesc,CrFwGetCurrentTime());
			cmpSpecificData->serialize(smDesc);
			CrFwOutStreamSend(outStream, cmpSpecificData->pckt);
			isRepeat = cmpSpecificData->isRepeat(smDesc);
		}
		if (isRepeat == 1)
			cmpData->outcome = 1;
		else
			cmpData->outcome = 2;
	}
}
```
Here, the problem is that nothing is done when isReady is false, This function should be modified as follows:
```
void PendingEntryAction(FwSmDesc_t smDesc) {
        . . . 
	if (isReady) {
            . . . 
	} else
		cmpData->outcome = 1;     /* New code */
}
```
Could Christian check whether the above change fixes the problem? If this is confirmed, I will then implement it in the next release of the framework.

reimers, 2017-10-31:
yes, just tested, and it solves the problem

closed 2019-07-31 by pasetti