pnp-software/cordetfw #113: Destination of Event Reports

pasetti, 2017-12-30:
We should consider making the destination of event report the same as the source of the command which enables them.

SergeValera, 2018-01-25:
Please, acknowledge that the Standard requirement 6.5.4b states: 
"The destination of the event reports generated by an event reporting subservice shall be declared when specifying that subservice".

pasetti, 2018-01-25:
Our current statement of compliance to requirement 6.5.4b (see appendix D of our [PUS Extension Document](https://github.com/pnp-software/cordetfw/blob/pus_ext/doc/pus/PusExtension.pdf) simply leaves the declaration of the destination of an event report to an application, In this ticket we are proposing to change this policy and use an automatic mechanism whereby the destination of an event report is the same as the source of the last command which enabled that report. 

The question is: would such a rule violate clause 6.5.4b? In my view, there is no violation because we are declaring the destination of an event report implicitly by defining the rule through which that destination is determined.

pasetti, 2018-03-25:
I have defined the destination of event reports as follows: "An initial destination is pre-defined. At run-time, the destination is the same as the source of the most recent command which enabled the event report". The implementation must be adapted.

pasetti, 2019-03-15:
Update has been made in release 0.2 of the PUS Extension Specification.

closed 2019-03-15 by pasetti