pnp-software/cordetfw #116: Check on N Parameter in Commands

pasetti, 2018-01-04:
Incoming commands which carry a group of identical instructions also carry an "N Parameter" which determines the cardinality of the ground. The value of N must be consistent with the command's length. This check should be added to the default configuration check for incoming commands.

SergeValera, 2018-01-25:
The N effectively represents the number of instructions carried within a request. Some instructions have a fixed size, e.g. TC[2,1] so, checking on-board the consistency of the N value with the length contained within the header of the CCSDS TC packet is possible. However, not all types of instructions imply that the related instructions have all the same size. This is for example the case of TC[21,1]. As such, checking the consistency with the length is not really adequate. However, during the extraction of each instruction, verifying that there is still place enough in the packet is adequate...

pasetti, 2019-03-30:
We check the CRC of a packet which verifies that the packet has not been corrupted and, for the rest. we should rely on the ground having checked the value of the command parameters. Applications which wish to have such a check can introduce it indirectly by introducing a check in the packet getter function such that they never get a parameter beyond the end of the packet.

closed 2019-03-30 by pasetti